CATIA-Systems/Modelica-Arduino #2: Support for Visual Studio 2013/Visual C++ 2013 Express Edition (12.0) or newer

lvanfretti, 2017-07-23:
Hi @t-sommer 

I've been trying to find an installer for Visual Studio 2012, however, it is no longer available in the distribution website: https://www.visualstudio.com/downloads/
I've searched around, but was not able to get the right installers.

I was wondering if you could help me get the version required by your project, or if you can provide support for VS2013 or newer (VS 2017 is available in the link above, for example).
In Dymola 2018 FD01 I have a working version of Visual Studio 2013 (compiler verified) and a non-functioning version of Visual Studio 2015 (that I guess I could get working somehow).

Thanks for any help!

t-sommer, 2017-07-24:
With the next release we're moving to VS 2015. You can try it out by cloning or downloading the latest version of the develop branch.

lvanfretti, 2017-07-24:
Thanks! I should have dogged through the other branches before bothering you!
This is a really nice teaching example, I enjoyed seeing the hardware version in Prague- it might be useful to share the HW specs in future versions.

closed 2017-07-24 by lvanfretti

lvanfretti, 2017-07-25:
@t-sommer I just tested successfully both "Blink" and "Fade" examples using Dymola 2018 and VC2015.
Thanks!
[image]