CATIA-Systems/Modelica-Arduino #7: Compiler issue

marcbudinger, 2020-04-08:
Dear T-Sommer,

I try to use your Arduino package. I have Dymola v2020x and express compilers 2014,2015,2017.
I'm not able to simulate the blink example. I have the following feedback error:
"
Compiling and linking the model (Visual C++). 
 dsmodel.c
LINK : fatal error LNK1181: impossible d'ouvrir le fichier en entrée 'ModelicaArduino.lib'
 Error generating Dymosim. 
"
I have unziped arduino into the Dymola directory.
Do you have an idea of the source of the error ?

Thank you, Marc

PS: I use also your FMPy package with more success. Thank you  for your work.

t-sommer, 2020-04-08:
Did you follow the steps in https://github.com/CATIA-Systems/Modelica-Arduino#getting-started ?

ThomasCastellano, 2021-12-12:
Hello,

I have follow all the steps you give on the main page but I get the exact same issue with the latest version of Dymola (3DExpérience Behavior Modeling) and the Visual Studio 2015 compiler (same thing with the 2022 compiler).

I am currently working on my semester project and I need your package to simulate an Arduino. 

Please, can you help me ?

Thank you,
Thomas

[image]

t-sommer, 2021-12-14:
I cannot reproduce the problem with Dymola 2022 and Visual Studio 2015.

ThomasCastellano, 2021-12-15:
Thank you for still following this thread.

I just tried on my desktop with VS2019 and I don't have this issue anymore.

However, it seems that your library is not compatible with the latest MSL 4.0.0 because some MSL's packages have been renamed (Modelica.SIunits is now Modelica.Units.SI for instance). 

[image]

I will try to install MSL 3.2.2, but having a student account I don't have the permissions for doing it myself.

I don't want to bother you, but could it be possible to have a compatibility update of your library ?

Thank you,
Thomas

t-sommer, 2021-12-16:
You should be able to automatically convert the library to MSL 4.0.0 with the dialog that shows up when you load the library in a newer Dymola release. Changing the dependency to MSL 4.* would cut support for older Dymola versions.

closed 2021-12-16 by t-sommer

ThomasCastellano, 2021-12-16:
Thank you for your answer.
However, I already tried this but that didn't fix the issue.

stephandiehl, 2021-12-20:
You have to update the arduino library to MSL 4.0. This could be done automatically by executing the conversion scripts. After importing the library in 3DEXPERIENCE, please press the second button in "Possible Actions".

[image]