CATIA-Systems/Modelica-Arduino #19: Error with Arduino.Examples.Blink

MMNDOUR, 2023-07-04:
Hello 
I tried to test the arduino library in open modelica. however I had problems. 
I followed the tutorial on this link https://github.com/CATIA-Systems/Modelica-Arduino
I can't figure out if it comes from the compiler. I searched with the errors obtained, however there are few elements apart from the tutorial on github  
Any help would be appreciated.
System:
Connected to OpenModelica v1.20.0 (64-bit)
Connected to OMSimulator v2.1.1.post188-gaf996ad-mingw
Connected to OpenModelica v1.20.0 (64-bit) built for Microsoft Windows [Version 10.0.19044.2251].
The running OS is Windows 10 Version 2009 on x86_64.

Warnings 
[1] 14:24:23 Ecriture Avertissement
[Modelica_DeviceDrivers: 9:16-9:38]: Conversion-annotation contains unknown element: nonFromVersion.

Errors 
[1] 14:34:29 Traduction Erreur
Variable run not found in scope <global scope>.

[7] 14:59:06 Traduction Erreur
[Arduino.Internal.buildSketchOM: 21:3-21:18]: Function remove not found in scope buildSketchOM.

C:\PROGRA~1\OPENMO~1.0-6\tools\msys\mingw64\bin\ld�: ne peut trouver -lModelicaArduino
clang: error: linker command failed with exit code 1 (use -v to see invocation)
mingw32-make: *** [Blink.makefile:37: omc_main_target] Error 1
mingw32-make: Leaving directory 'C:/Users/mymza/AppData/Local/Temp/OPENMO~1/OMEdit/ARDUIN~1.BLI'
Compilation process failed. Exited with code 0x00000002.

[image]

MMNDOUR, 2023-07-04:
[image]

t-sommer, 2023-07-10:
Did you use the [latest release](https://github.com/CATIA-Systems/Modelica-Arduino/releases/download/v0.3.0/Modelica-Arduino-0.3.0.zip) or source code?

closed 2023-08-03 by MMNDOUR